deividAlfa/stm32_soldering_iron_controller #3: Bugs

ALFURRER, 2021-08-19:
Hello, I am facing a problem. When trying to calibrate the soldering iron tip, this bug appears with the image! 
[image]
[image]

ALFURRER, 2021-08-19:
SW: 2021-08-18
HW:KSGER V3.x

deividAlfa, 2021-08-19:
Have you removed the screen at some moment?
That's usually caused when you change the screen and some connection gets damaged.

ALFURRER, 2021-08-19:
No, I did not shoot the screen, there were no such problems on the previous firmware version from 08-10-2021. Yes, I changed the op amp to a precision one.
Again, this has not happened since the previous version of the firmware. And I didn't make a backup.

ALFURRER, 2021-08-19:
Again, this happens if I just turn the encoder knob in the main menu

ALFURRER, 2021-08-19:
[image]

deividAlfa, 2021-08-19:
Strange.
You can download any older version by checking the file history.
Click the 3 dots ...  view file. As far as know no changes were made to the screen, but I'll check that again.
https://github.com/deividAlfa/stm32_soldering_iron_controller/commits/master/BOARDS/KSGER/%5Bv3%5D/STM32SolderingStation.bin

ALFURRER, 2021-08-19:
Ok) thanks)))

ALFURRER, 2021-08-19:
I will definitely check and inform you

ALFURRER, 2021-08-19:
I have checked the old firmware version from 10-08-2021. Everything works, the soldering iron tip is calibrated.
Once there was a bug with the display. I think that this bug appears in the event that it is bad to clear the memory of the controller before flashing.
Now I will try to fill in the new firmware from 08-18-2021.
I will unsubscribe based on the results.
PS. Sorry, my English is very bad, I use a translator

ALFURRER, 2021-08-19:
Я прошил плату новой версией прошивки. Откалибровал жало паяльника. Все в норме. Больше багов с экраном не наблюдал.
Скорее всего, этот баг появляется при плохом стирании памяти контроллера перед заливкой новой версии прошивки. 
Благодарю вас за помощь и консультацию, прошу меня извинить за беспокойство

ALFURRER, 2021-08-19:
I have flashed the board with a new firmware version. I calibrated the soldering iron tip. Everything is okay. I did not see any more bugs with the screen.
Most likely, this bug appears when the controller memory is badly erased before uploading a new firmware version.
Thank you for your help and advice, I ask you to excuse me for worrying.

deividAlfa, 2021-08-19:
I had this issue caused by a bad earth. My board didn't like much joining Earth (AC Plug ground) and Gnd (Board negative) together, neither leaving them opened.
I don't have a good earth at home, so it caused interferences, touching anything caused strange behaviors, the controller crashed, the readings were crazy and also the screen failed some times.
Connnecting earth and gnd through a resistor (20K-100K) works nicely for me.

ALFURRER, 2021-08-19:
I also do not have a normal grounding at home. But in any case, I'll see how it works in the future. If anything, I'll let you know. Thank you for your help

closed 2021-08-23 by deividAlfa