deividAlfa/stm32_soldering_iron_controller #6: Shake wake sensitivity too high

zyonee, 2021-09-10:
Hi!

I'd like my device to stay in sleep mode until I actually move the soldering pen, so I set it to "Wake mode: SHAKE". This works, but it works a bit too well. If I just nudge my desk very gently, the sleep is interrupted. Is there any way to decrease the shake sensitivity? Maybe some debounce logic could help?

deividAlfa, 2021-09-10:
Set Shake to Standby, and button to All.
That way only the button will wake from sleep.

deividAlfa, 2021-09-10:
As the name suggest, it detects changes, not states.
When you move the handle it will generate hundreds of small pulses. 
If your handle is so sensitive, just set Shake to Standby, and button to All, that way only the button will wake It up from sleep.

deividAlfa, 2021-09-10:
If your handle is so sensitive, just set Shake to Standby, or disable it, leaving the button for that.
The Shake input will be used anyways when the station is running, keeping it from entering low power modes, but once it entered sleep/standby, it won't wake it up.

zyonee, 2021-09-10:
Thanks for the feedback! I'll look into getting a mercury switch. Thanks for this great firmware, keep it up!

deividAlfa, 2021-09-12:
I added a sensitivity workaround. Will add an option to enable to disable it.

zyonee, 2021-09-12:
Awesome! Thank you!

zyonee, 2021-09-12:
I've played around with the minimum and maximum tick values. Seems like your default is still too sensitive for my device. Changing it to <200 || >400  seems to be golden for me. I have to tap my desk in the exact right rhythm to exit sleep/standby, and if I pick up my soldering pen, all it needs is a bit of a shake (which is what shake wake implies). Good work!

deividAlfa, 2021-09-13:
Nice! Closing this then.

closed 2021-09-13 by deividAlfa