deividAlfa/stm32_soldering_iron_controller #14: Проблема с энкодером на KSGER v3.x

Alex-TV, 2021-12-14:
GhbПри прокручивании энкодера в одну сторону, курсор в меню сначала идет в верном направлении, после меняет направление. Также это замечено при изменении любых параметров. Если вращать медленно по одному щелчку, проблемы нет. Но стоит немного увеличить скорость вращения проблема повторяется.

deividAlfa, 2021-12-14:
Is the original fw working correctly?
Nobody has complained about this, ever.
Might be a defective rotary encoder? Resolder the stm32 pins.

Alex-TV, 2021-12-14:
Worked correctly. On old firmware. I'll check the contacts stm32. Zero calibration is also not clear. Incorrect NTC temperature. We managed to establish the correct temperature, only the selection of resistance in the settings. Did I load the firmware in the STM32 incorrectly? Didn't clear more than 64Kb memory.

deividAlfa, 2021-12-14:
But you have to follow the instructions. It clearly states that you shgould clear the first 126KB.
If the programming didn't fail, I guess it was OK.
For the zero calibration, check the operation manual, it's pretty straightforward.
If your board has a different NTC or pullup resistor, report that. The firmware is adjusted to what users have reported, I don't have any Ksger myself.

Alex-TV, 2021-12-14:
There is a 4.7K on  3.3V pullup resistor. After the resistor 4.7 kOhm goes to STM.

Alex-TV, 2021-12-14:
Thanks, I figured it out. The Chinese pen contains a 15 kΩ thermistor. In the native 10kOhm. After soldering, the encoder started working. Although sometimes there are refunds.

deividAlfa, 2021-12-15:
You have to measure the thermistor at 25°C.
Resolder all stm32 pins to ensure...
Sometimes there're also shorts, all those problems come from the bad quality at the factory...

closed 2021-12-15 by deividAlfa