deividAlfa/stm32_soldering_iron_controller #15: Different Chip on the T12 board. Will it be compatible?

benjaminjacobreji, 2022-02-02:
Hi

I got my KSGER T12 Mini earlier last month.
Used it for a bit (very light workload aka testing on wires, twice) and then wanted to update the software and came accross this project.
So got myself a ST-Link and opened up the T12 to check for pins and which chipset was on it.

Imgur Link: [Ksger T12](https://imgur.com/a/oGGiVJx)

I found the chipset labelled "ARM MM32SPIN 27PF 76508-p5 G3H6 016" and not a STM32.
I tried googling, but couldn't find any useful information on the chipset.

Will this project be compatible, if so which firmware image should I flash?

Have a great day xD

Warm Regards
Benjamin

benjaminjacobreji, 2022-02-02:
The only information I could find was on AliExpress here: [AliExpress KSGER T12]

(https://www.aliexpress.com/item/4000474710749.html#:~:text=Note%3A%20We%20have%20changed%20the%20soldering%20station%20chip%20to%20MM32%20from%202022/12/1.%20But%20it%20will%20be%20same%20firmware%20and%20same%20function%20with%20STM32%2Cso%20please%20donot%20worry%20about%20it.)
"Note: We have changed the soldering station chip to MM32 from 2022/12/1. But it will be same firmware and same function with STM32,so please donot worry about it."

benjaminjacobreji, 2022-02-02:
I saw this update regarding the new chips from DavidAlfa over on [EEVBlog](https://www.eevblog.com/forum/reviews/stm32-oled-digital-soldering-station-for-t12-handle/msg3970565/#msg3970565) and as stated I understand the immense work that needs to go in for porting and why it isn't feasible to do for a single board.
 
As a reference for future searchers, I won't delete the issue (unless instructed otherwise) but rather close it.
This way people searching for "MM32SPIN KSGER T12" will have some place with relevant information.

Warm Regards

closed 2022-02-02 by benjaminjacobreji

deividAlfa, 2022-02-02:
Hello, yes, exactly as described in eevblog.
First of all, ST-Link doesn't work with it, at least out of the box.
And it's a M0 chip, while Ksger fw uses stm32 F1 series (Cortex M3).
Replace it with any STM32F101/102/103  C8/CB and it will be fine.
GD32 clones work well, too.

Regards