deividAlfa/stm32_soldering_iron_controller #22: Iron Min Max limits always Celsius

TheStrath, 2022-06-11:
iron_screen.c does not consider tempUnit setting when limiting min and max value. May be others

deividAlfa, 2022-06-12:
The temperature unit can be ony changed in system_screen.c, iron_screen.c takes the current setting.
Wwhy is it needed? Why are you reporting *no* issue at all?

TheStrath, 2022-06-12:
Steps to recreate one instance of the issue:
System: Temperature: Set to degrees C
Iron: Max Temp: set to 480 degrees C
System:  Temperature: Set to degrees F
Iron: Max temp display reads 900 degrees F
Adjust value, Max limited to 480 degrees F

Confirmation:
System: Temperature: Set back to degrees C
Iron: Max temp display reads 250 degrees C

deividAlfa, 2022-06-12:
Thanks. Now it's a proper bug report, explaining the issue itself.
Will check it and update soon.

deividAlfa, 2022-06-12:
Fixed. Use the binaries from Release v1.0:
https://github.com/deividAlfa/stm32_soldering_iron_controller/releases/tag/v1.0

closed 2022-06-12 by deividAlfa