deividAlfa/stm32_soldering_iron_controller #25: Issues upgrading using STM32CubeProgrammer

fiechr, 2022-07-15:
I have a KSGER 2.0 and 3.0 on which I've successfully installed this great firmware aprox. half a year ago.

Today, I wanted to upgrade them to the latest release. But I'm having issues uploading the firmware using Linux (Manjaro). It's the same hard- and software I used for the initial install (read/write protection is still disabled).

I've also used this for different other STM32 MCUs, so I'm pretty sure the setup should be fine.

Still, this time I get an error (see last line below).

I wanted to keep my calibration data, so I've skipped erasing all data. I did NOT try it with erasing all data, yet. Wanted to ask first if there is any idea what to do?

[image]

```
12:06:46:555 : ST-LINK SN  : 48FF6E066770545657271967
12:06:46:555 : ST-LINK FW  : V2J39S7
12:06:46:557 : Board       : --
12:06:46:559 : Voltage     : 3,27V
12:06:46:576 : SWD freq    : 4000 KHz
12:06:46:576 : Connect mode: Normal
12:06:46:576 : Reset mode  : Software reset
12:06:46:576 : Device ID   : 0x410
12:06:46:576 : Revision ID : Rev X
12:06:46:588 : flash loader /opt/stm32cubeprog/bin/FlashLoader/0x410.stldr is loaded
12:06:46:588 : Debug in Low Power mode is not supported for this device.
12:06:46:589 : Reading data...
12:06:46:589 : r ap 0 @0x1FFFF7E0 0x00000004 bytes Data 0xFFFF0040
12:06:46:710 : UPLOADING OPTION BYTES DATA ...
12:06:46:711 :   Bank          : 0x00
12:06:46:711 :   Address       : 0x4002201c
12:06:46:711 :   Size          : 8 Bytes
12:06:46:711 : Reading data...
12:06:46:711 : r ap 0 @0x4002201C 0x00000008 bytes Data 0x03FFFFFC
12:06:46:711 :   Bank          : 0x01
12:06:46:711 :   Address       : 0x1ffff800
12:06:46:711 :   Size          : 16 Bytes
12:06:46:711 : Reading data...
12:06:46:711 : r ap 0 @0x1FFFF800 0x00000010 bytes Data 0x00FF5AA5
12:06:46:711 : UPLOADING OPTION BYTES DATA ...
12:06:46:711 :   Bank          : 0x00
12:06:46:711 :   Address       : 0x4002201c
12:06:46:711 :   Size          : 8 Bytes
12:06:46:711 : Reading data...
12:06:46:711 : r ap 0 @0x4002201C 0x00000008 bytes Data 0x03FFFFFC
12:06:46:711 :   Bank          : 0x01
12:06:46:711 :   Address       : 0x1ffff800
12:06:46:711 :   Size          : 16 Bytes
12:06:46:711 : Reading data...
12:06:46:711 : r ap 0 @0x1FFFF800 0x00000010 bytes Data 0x00FF5AA5
12:06:46:727 : UPLOADING ...
12:06:46:727 :   Size          : 1024 Bytes
12:06:46:727 :   Address       : 0x8000000
12:06:46:727 : Read progress:
12:06:46:727 : Reading data...
12:06:46:732 : r ap 0 @0x08000000 0x00000400 bytes Data 0x20002800
12:06:46:732 : Data read successfully
12:06:46:732 : Time elapsed during the read operation is: 00:00:00.006
12:07:02:859 : Memory Programming ...
12:07:02:859 : Opening and parsing file: KSGER_V3_OLED.bin
12:07:02:859 :   File          : KSGER_V3_OLED.bin
12:07:02:860 :   Size          : 85,51 KB 
12:07:02:860 :   Address       : 0x08000000 
12:07:02:860 : Download in Progress:
12:07:02:862 :   Size          : 87564 Bytes
12:07:02:862 :   Address       : 0x08000000 
12:07:02:862 : Buffer program...
12:07:02:862 : halt ap 0 
12:07:02:863 : w ap 0 reg 15 PC   (0x20000000)  
12:07:02:863 : w ap 0 reg 17 MSP  (0x20000500)  
12:07:02:863 : w ap 0 reg 16 xPSR (0x01000000)  
12:07:02:866 : w ap 0 @0x20000960 0x00000200 bytes Data 0x00000000
12:07:02:867 : w ap 0 @0x20000000 0x00000004 bytes Data 0x0000BE00
12:07:02:885 : w ap 0 @0x20000004 0x0000053E bytes Data 0xF000B580
12:07:02:885 : Loader write range...
12:07:02:930 : w ap 0 @0x20000960 0x00002200 bytes Data 0x20002800
12:07:02:930 : W B1 in RAM @0x20000960 size 0x00002200 : 0058ms
12:07:02:930 : Init flashloader...
12:07:02:932 : halt ap 0 
12:07:02:932 : w ap 0 reg 0 R0   0x00000001
12:07:02:932 : w ap 0 reg 1 R1   0x00000000
12:07:02:932 : w ap 0 reg 2 R2   0x00000000
12:07:02:933 : w ap 0 reg 3 R3   0x00000000
12:07:02:933 : w ap 0 reg 4 R4   0x00000000
12:07:02:933 : w ap 0 reg 5 R5   0x00000000
12:07:02:934 : w ap 0 reg 6 R6   0x00000000
12:07:02:934 : w ap 0 reg 7 R7   0x00000000
12:07:02:935 : w ap 0 reg 8 R8   0x00000000
12:07:02:935 : w ap 0 reg 9 R9   0x00000000
12:07:02:935 : w ap 0 reg 10 R10  0x00000000
12:07:02:936 : w ap 0 reg 11 R11  0x00000000
12:07:02:938 : w ap 0 reg 12 R12  0x00000000
12:07:02:938 : w ap 0 reg 13 SP   0x00000000
12:07:02:938 : w ap 0 reg 14 LR   0x20000001
12:07:02:938 : w ap 0 reg 15 PC   0x20000005
12:07:02:939 : w ap 0 reg 16 xPSR 0x01000000
12:07:02:953 : w ap 0 reg 17 MSP  0x2000093E
12:07:02:954 : w ap 0 reg 18 PSP  0x00000000
12:07:02:954 : run ap 0 
12:07:02:955 : halt ap 0 
12:07:02:955 : r ap 0 reg 0 R0   0x00000001
12:07:02:955 : w ap 0 reg 0 R0   0x08000000
12:07:02:955 : w ap 0 reg 1 R1   0x00002200
12:07:02:955 : w ap 0 reg 2 R2   0x20000960
12:07:02:955 : w ap 0 reg 3 R3   0x00000002
12:07:02:955 : w ap 0 reg 4 R4   0x00000000
12:07:02:955 : w ap 0 reg 5 R5   0x00000000
12:07:02:955 : w ap 0 reg 6 R6   0x00000000
12:07:02:955 : w ap 0 reg 7 R7   0x00000000
12:07:02:955 : w ap 0 reg 8 R8   0x00000000
12:07:02:955 : w ap 0 reg 9 R9   0x00000000
12:07:02:955 : w ap 0 reg 10 R10  0x00000000
12:07:02:955 : w ap 0 reg 11 R11  0x00000000
12:07:02:955 : w ap 0 reg 12 R12  0x00000000
12:07:02:955 : w ap 0 reg 13 SP   0x00000000
12:07:02:955 : w ap 0 reg 14 LR   0x20000001
12:07:02:955 : w ap 0 reg 15 PC   0x2000002F
12:07:02:955 : w ap 0 reg 16 xPSR 0x01000000
12:07:02:955 : w ap 0 reg 17 MSP  0x2000093E
12:07:02:955 : w ap 0 reg 18 PSP  0x00000000
12:07:02:955 : run ap 0 
12:07:03:006 : w ap 0 @0x20002B60 0x00002200 bytes Data 0xFEC8F002
12:07:03:006 : W B2 in RAM @0x20002B60 size 0x00002200: 0075ms
12:07:03:007 : r ap 0 reg 0 R0   0x00000000
12:07:03:007 : halt ap 0 
12:07:03:007 : w ap 0 reg 15 PC   (0x20000000)  
12:07:03:008 : w ap 0 reg 17 MSP  (0x20000500)  
12:07:03:008 : w ap 0 reg 16 xPSR (0x01000000)  
12:07:03:009 : Loader write range...
12:07:03:067 : w ap 0 @0x20000960 0x00002200 bytes Data 0x20002800
12:07:03:068 : W B1 in RAM @0x20000960 size 0x00002200 : 0058ms
12:07:03:071 : halt ap 0 
12:07:03:072 : r ap 0 reg 0 R0   0x00000000
12:07:03:073 : Error: failed to download Segment[0]
12:07:03:073 : Error: failed to download the File
```

fiechr, 2022-07-15:
I've unchecked the "Skip flash erase before programming" and then it works without issues. Any idea why?

I'll wait before erasing the other station too. ;-)

deividAlfa, 2022-07-16:
No idea. This is no ST support...
I use classic stlink utility, cube prog is slow and buggy...

closed 2022-07-16 by deividAlfa