deividAlfa/stm32_soldering_iron_controller #31: Unknown Board rev. v2 / v3 mixup

kinoX51, 2022-08-22:
I've just ordered from the "KSGER Official Store" of AE an "KSGER STM32 V 3,1"
Unfortunately it came delivered with an MM32Spin, which is incompatible with your custom firmware.

I came across your eevblog-forum post (https://www.eevblog.com/forum/reviews/stm32-oled-digital-soldering-station-for-t12-handle/msg4286710/#msg4286710), that it is possible to replace the mm32 with an genuine stm32.

My problem is, that I can't tell which version of solderingstation I got.
6 Pin means OLED but the controlboards hast v.20 printed on it and there is no V2 OLED-Version of the firmware.

[image]
[image]

deividAlfa, 2022-08-22:
Hi, looks like a v3... Slightly modified, but same layout, ksger does this all the time!

kinoX51, 2022-08-22:
ok, thanks! Do you think it's safe to flash v3 or can I help you with anything to verify that?

deividAlfa, 2022-08-22:
There's nothing that could go wrong if using the wrong firmware, it just won't work.

deividAlfa, 2022-08-24:
Closing, re-open if having any issues.

closed 2022-08-24 by deividAlfa