deividAlfa/stm32_soldering_iron_controller #45: Display support for T12-959 2.6 inch OLED screen (after MCU swap)

mkbehbeh, 2023-02-18:
I have the T12-959 with the larger screen.  The stock firmware leaves a lot to be desired.  The stations works similar to the old analog 888 where you set the temp with the dial (it does have sleep at least).  I am comfortable replacing the STC32 with a STM32 but I could not find if the firmware would support the larger display.  It is the 2.6 inch display.,  Most sites describe it as an OLED but a few describe it as an LCD.  It seems to only have red/white color.

Can you clarify that the display will work if the STC chip is replaced with an STM32?

Thanks,

deividAlfa, 2023-02-18:
Hi, first of all you'll hardly replace the stc with a stm32, it's a completely different chip, not a clone!
There's no stm32 you could directly replace it with.
The display is LCD almost for sure, a 2.6" oled would be too expensive, but I know anything about it.

mkbehbeh, 2023-02-20:
Thank you for the information.  Looks like the 959 is a poor choice if you want options for the firmware

closed 2023-02-20 by mkbehbeh