deividAlfa/stm32_soldering_iron_controller #46: Cooperation with IronOS project

violavon, 2023-02-18:
Would cooperation with [IronOS project](https://github.com/Ralim/IronOS) be possible?

deividAlfa, 2023-02-18:
How could I contribute?
I only do C, I have zero skills at Python, C++, etc...
Also I barely have any free time lately, in fact I haven't turned the computer for an entire month now! 😅

violavon, 2023-02-19:
Well, I think it's a good idea to reduce reinventing a wheel. If these two projects can share code with each other (or even merge into one; just an idea) that would be great. More active contributors around open-source firmware for soldering irons.
Great, IronOS is also mainly written with C!
It's ok, you aren't the only one. 😁 Btw, thanks for your work on this firmware so far, it's amazing!
Yup, although maybe user interface requires some changes. biarss? You mean boards?
Thanks.
All controllers? Could you clarify?
Good point.

IronOS is new to you? I think you will like it. I've been using it for couple or three years now with Miniware TS100 and a year with MHP30, and I love it! Although the lack of USB-C with TS100 bothers me, so maybe I'll upgrade to TS80P or many are [excited about Pine64 Pinecils](https://www.theverge.com/23541478/pinecil-best-cheap-soldering-iron-open-source), not sure yet.

violavon, 2023-02-19:
Mostly now I'm inquiring about the interest of these two projects in cooperation. I believe that any cooperation is beneficial and all ideas are welcome!

deividAlfa, 2023-02-20:
I can't see any possible cooperation because:
- The targetted devices are completely different.
- I consider this project done.
- I have little to no time, everyday life has priority over some cheap chinese stations.
- Lost interest long time ago.

The code is there, take it as you wish! 👍

closed 2023-02-20 by deividAlfa

violavon, 2023-02-20:
> The targetted devices are completely different.

Completely different? The difference between soldering irons and stations isn't huge. Of course, UI needs work, but it isn't difficult to maintain two UIs in one firmware and I'm sure we'll come up with something that works almost the same in both use cases. It just needs dev and design. Besides, we haven't thought of ways to cooperate yet. It could be anything e.g. developing driver compatibility between IronOS and this firmware.

> I consider this project done.

What do you mean by done? Not anymore new features or support for new devices?

> I have little to no time, everyday life has priority over some cheap chinese stations.

Totally understandable.

> Lost interest long time ago.

For the whole project? One of my suggestions was that merging the projects would be an option, but I didn't know the situation is this. It's always nice to have several softwares to choose from. Well, if you're not going to continue this project, would it be good to add the best parts of this to IronOS? Then the work wouldn't be wasted.

> The code is there, take it as you wish! +1

I know, but it's a great job of you and that's why I think it's important to ask you first. And I would really appreciate it if you could help us a little with this, at the beginning.

deividAlfa, 2023-02-20:
No need to over-complicate things, Iron OS is probably more than enough for these! 👍
I developed this to keep myself busy during the pandemic, as neither I was able to work due a surgery problem, so I was 24/7 at home, alone, so I had to do something to not get completely crazy 🤪, but now I no longer have no time for that kind of stuff!

> The difference between soldering irons and stations isn't huge
The main difference is a very specific hardware target.
My firmware needs a lot of options to properly work with all Ksgers, Queecos, etc, also being a station it can work in different ways.
The TS** soldering irons are way simpler, they simply need to heat up and work, 85% of the options in this firmware are not needed at all.
Not forgetting this firmware is designed for 128x64 oleds, everything would need to be redesigned!
There's no need to overcomplicate things, right?

violavon, 2023-02-21:
> Iron OS is probably more than enough for these! +1
 
For these?

> The main difference is a very specific hardware target.
My firmware needs a lot of options to properly work with all Ksgers, Queecos, etc, also being a station it can work in different ways.
Probably 85% of this firmware options are not needed at all for the TS soldering irons.
Not forgetting this firmware is designed for 128x64 oleds, everything would need to be redesigned!
There's no need to overcomplicate things, right?

I think it's vice versa; IronOS is more expansive. For example, IronOS has [33 settings](https://github.com/Ralim/IronOS/blob/dev/source/Core/Src/settingsGUI.cpp#L97) and this firmware has 31. How do you think this firmware is more expansive than IronOS?

TODO:
 * Controls: IronOS is currently typically run on the devices with two buttons and this firmware on the devices with knob (roll and button). I think quite easy to solve. In IronOS the right button works like the roll in this firmware and the left button like the knob's button.
* Display: IronOS currently typically run on the devices with a small rectangular oled display. This is more complicated, we need to either make it responsive or make our own user interface for small and large displays. Responsive isn't necessarily difficult. Almost all IronOS's setting menus can quite logically be spread over large display, every line below another.
 * UI: New settings can add to among of the old ones. When the previous points are solved, remains is mostly the front page/windows. I suggest leaving two different the front windows, but making uniform style (use the same font, icons, etc. in both) for them.
* Drives: Porting these isn't a problem. IronOS already support STM32 and this is the easiest of these.

Besides, it's good to remember that IronOS will anyway add support for stations at some point, so...