deividAlfa/stm32_soldering_iron_controller #47: Rename master branch to main

violavon, 2023-02-19:
Rename master branch to main.

> For the longest time, the default branch in most Git repositories was named "master". Fortunately, many people have become aware that this terminology (even more evident in "master/slave") should be replaced! The tech industry should move to a more inclusive, open culture - and removing language like "master/slave" is an important step in this journey.
> 
> In the public discussion, a handful of different alternatives for "master" have popped up, "default" and "primary" being some of them. But the most popular term seems to be "main".

https://www.git-tower.com/learn/git/faq/git-rename-master-to-main

fourtwizzy, 2023-02-20:
What an amazing change you have proposed. Maybe we can do the following as well.

Rename MasterClass.com to PrimaryClass.com
Change MASTERbate to DefaultBate
Strip all titles from chess Grand Masters and rename them Grand Chess Persons
Rename Master Splinter of Teenage Mutant Ninja Turtles to Primary Rat Persons
Change Master's Degrees to Primary degrees. Wait that won't work for all the liberal arts majors.

Please keep your DEI, Woke BS to yourself.

violavon, 2023-02-20:
Maybe you could go somewhere else to blast… I only suggested rename master branch to main because I think that is a better name for it and more common these days.

fourtwizzy, 2023-02-20:
> Maybe you could go somewhere else to blast… I only suggested rename master branch to main because I think that is a better name for it and more common

Maybe you can provide a complete list of other words that offend you. Any other words you need banned or changed? 

Sorry but stupid needs to be called out where possible, especially from someone who has been on GitHub for 3 days. Here are a few major projects with Master branches. Maybe you should continue your social justice warrior campaign. Go back to sleep, your woke blast is trash.

https://github.com/996icu/996.ICU
https://github.com/kamranahmedse/developer-roadmap
https://github.com/donnemartin/system-design-primer
https://github.com/tensorflow/tensorflow

Any more of you id10ts and the dictionary is going to be 1 page.

fourtwizzy, 2023-02-20:
Also you've had an account for three days. You don't think it is a better name, you just want to go pushing some agenda that doesn't need pushing. I imagine you look to see the branch name just to rock your jollys. What on earth will you do, when people stop accepting your version of fascism and no longer allow the mental patients to run the institution?

deividAlfa, 2023-02-20:
Do you spend hours in the Internet searching offending connotations in everything?
Sorry, nope, there's nothing wrong in using the word Master, wokes need to wake up.
At least I has some serious laughts at @fourtwizzy 😂.
Closed!

closed 2023-02-20 by deividAlfa

violavon, 2023-02-20:
At least half of my local repos still uses master branch. However, since so many projects use main these days, I've become familiar with it and it's also clear that the master isn't coming back. Sure, master is still used in some projects, but it seems that main is more common. These were my reasons for suggesting this, nothing more or less.
Really unfortunate that the reception is this. ☹️ Tbh maybe this says more about your problems than mine.

P.S. Yeah, my GitHub account is fresh because I use local repo/Gitea. This week I found projects here that I want to contribute in.

deividAlfa, 2023-02-20:
I have nothing against naming the repo to "main", neither you.
It's the WokeTard in-culture what takes my insides out, people taking things out of context, feeling offended because anything randomly contains "Man", "Black", "Slave", they're usually useless, contributing ZERO to society, all they do is finding new things to complain about.
It's their problem if they instantly think in beaten African slaves. That was 200 years ago!
Total education failure. They do what they do because nobody educated them properly, saying "NO" when it was needed.

fourtwizzy, 2023-02-20:
> Sure, master is still used in some projects, but it seems that main is more common. These were my reasons for suggesting this, nothing more or less.

I find this comment to be disingenuous. We all understand the reasons why you are pushing for the idea, and none of them are that benign. This is pushing a garbage narrative and political agenda. There are songs that come out using the "hard R", still to this date in 2023. That is an awfully racist term. I am going to guess you aren't asking those musicians to change their language. 

See you and your woke "peers" have decided to dictate how things should be written or now in the case of Roald Dahl, how they should be rewritten. Instead you and your self righteous peers have decided that the laws regarding books in Florida are draconian and fascist, yet take look in the mirror. The guy was born in 1916, not 2016. Guess what, times were different. It doesn't mean you and your fascist army get to come in and dictate Roald Dahl's words. Ban the books like they did in Florida, or stop the whitewashing and censoring of everything.

I've come up with a list of things we should correct.

- The use of args and kwargs. Args and kwargs reminds me of the word argument, which reminds me of literal violence. Programmers need to put an end to this bigotry.
- Legacy hardware with a drive that explicitly states Master, Slave, or Cable Select should be powered off immediately. First the use of Master and Slave. However I think more importantly "Cable Select" should be removed in its' entirety. It could bring back trauma of children who weren't selected first for team sports in physical education.
- All of computer science must find new words for Bit, Byte, and Nibble. They may make a fat kid feel self conscious, because he took too many bytes, and ate a bit too much, when he really just needed a nibble at his size.
- Helper functions further push the narrative that some "main" function has the power to just boss around a "helper function". Literal violence, needs to be changed immediately. All functions deserve equality and equity, instead of just being used by main.
- Float variables, for the love of all that is holy. How insensitive can you be towards "non-swimming persons"?
- foo(bar)? I think not. This is as far from inclusion as you can get. There are plenty of alcoholics on their recovery journey that cannot (or should not) go to the bar. Have you no shame? Have you not thought of these other souls lost on their journey?
- The variable type Char should have been erased years ago. You are literally supporting racism. Char, char-broiled, which in turn could trigger someone with regards to lynching.
- Killing a process. Do I even need to mention how violent killing anything is. (Except fetuses, those are cool).
- Terminate a job? Termination of workers is a serious issue that should be dealt with. Terminate also rhymes with exterminate, which is what some chap tried to do from 1935 to 1945.
- A hung process. First I don't even want to know how they measured, but clearly this could cause self confidence issues to bubble to the surface. Why does the size of the process matter? It is what you do with the process that counts, not the size of it.
- You are asking for collaboration with IronOS, did you even stop to consider the word Iron. Like from the Iron ages. Think of all the atrocities that happened. The pre-Roman Iron Age started in 500 BC and ended in 1 AD. Do you not care about who the Romans killed then?! To some people that guy was the savior of humanity, and you are just going to allow for the installation of IronOS firmware on their devices? This is abhorrent behavior and I'm disgusted.
- Using whitelists and blacklists?! I don't even need to explain how terrible this is.
- Pointers! Why is anyone or anything pointing at anyone or anything?! People could seriously feel their social anxiety creep up when someone is pointing at them. Yet here we are just using pointers non-stop.

It is increasingly clear that you have not considered all the hateful words you use in your day to day life, and how they affect others. I will sit back and ponder while I look at my Master's degree, and question when will they call it something more appropriate like a "higher level learning degree", but then I remember that to be a higher level of learning there must be a lower level of learning. Therein making people with "lower level" degrees feeling worthless. To make matters worse, they literally call some of these lower level degrees "Bachelor's Degrees". Then I realized just look at the male patriarchy forcing the term bachelor on people. Bachelor is defined as a single male, and we all know the privileges that come with being a male. The misogyny of Bachelor's degrees is just palpable!

>Total education failure. They do what they do because nobody educated them properly, saying "NO" when it was needed.
They also needed to get spanked, not some safe space. Parents should have probably put terms and conditions on the "you can be anything you want when you grow up". Many are now finding that isn't the case, even with surgery.

And all of that combined, the woke id10ts of the world have become the fascists their great-grandparents or grandparents fought. The sad irony (sorry didn't mean to use such an offensive precious metal) of world affairs in 2023.

violavon, 2023-02-20:
All I can say is sit down and calm down. First of all, the reason for suggesting master -> main change is what I said, and I don't care if you believe it or not. I don't doubt that you still come across a lot of master, but I have come across and noticed that main is more common. This completely depends on which projects you work on.

And even though woke etc. wasn't behind my proposal, after these comments, I think your reaction just shows that woke is needed in appropriate amounts _for people like you_. I mean, when you explode and write a novel like this because of proposition like this, something is very wrong.

I won't spend any more of my time on this.

Thank you @deividAlfa for understanding after I explained my purpose.

fourtwizzy, 2023-02-20:
>All I can say is sit down and calm down.

Plenty calm, and I prefer a standing desk.

>I don't care if you believe it or not. 

And you shouldn't, still think you are lying though.

>I think your reaction just shows that woke is needed in appropriate amounts for people like you. 

Incorrect. No amount of "wokeness" is appropriate in any situation. There is nothing woke about wokeness, it is just disguised racism. Looking for racism where it doesn't exist.

>I mean, when you explode and write a novel like this because of proposition like this, something is very wrong.

If you consider that exploding I feel sorry for you. The real world is far harsher than my little novel there. That is what you call sarcasm. You can even freely take it as a joke, both at your expense and at the woke culture. 

You've already asked the chap from IronOS to change from Master to Main too. Ever hear the phrase "you can't BS a BSer"? I smell your BS, and it is rotten, to the core, just like the "main" agenda you want to push. Good luck mastering life friend. Reality is cold, harsh, and doesn't care what your preferred words are.