deividAlfa/stm32_soldering_iron_controller #49: iron profile auto detect

FrankHovis, 2023-03-11:
I notice there's a menu option to Set which iron profile (T12, C210, C245) is used. Is there any way for the firmware to detect which type of iron is plugged in? These controllers https://www.aliexpress.com/item/1005002380823696.html use a resistor to identify the iron used.

deividAlfa, 2023-03-11:
If this was custom hw, it would be pretty easy, but existing controllers don't have any way of detection.

Long time ago I thought on adding a resistor to the shake pin, but none are connected to ADC inputs.

Remember this project aims to replace the original fw, without heavy hw modifications, avoiding breaking original fw compatibility or risking ruining the board by unexperienced hands.

deividAlfa, 2023-03-11:
These controllers don't use NTC at the handle.
[image]

closed 2023-03-11 by deividAlfa