deividAlfa/stm32_soldering_iron_controller #50: OpenOCD: no flash bank found for address 0x08010000 (SOLVED)

chtaube, 2023-03-18:
I just ran into an issue where flashing the chip with OpenOCD fails with this message:

```
❯ openocd
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V3J8M3B5S1 (API v3) VID:PID 0483:3753
Info : Target voltage: 3.280266
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003340 msp: 0x20005000
Info : device id = 0x20036410
Info : flash size = 64kbytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003340 msp: 0x20005000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003340 msp: 0x20005000
** Programming Started **
Warn : no flash bank found for address 0x08010000
** Programming Finished **
** Verify Started **
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'script' 
at file "embedded:startup.tcl", line 26
in procedure 'program' called at file "openocd.cfg", line 13
in procedure 'program_error' called at file "embedded:startup.tcl", line 591
at file "embedded:startup.tcl", line 530
```

Same error message when trying to flash the stock firmware, so device was sort of bricked. But...

After some investigation I found OpenOCD's scripts for the stm32f1x platform were designed to support the "official" 64k flash size only and trying to flash larger program sizes results in the above message.

The OpenOCD documentation gives a hint how to resolve this: https://openocd.org/doc/html/Flash-Commands.html#index-stm32f1x

This forum post (in German language) has a practical solution that worked for me: https://www.mikrocontroller.net/articles/STM32F103C8T6_STM32_Billig_Board

What I did:

1. in `/usr/share/openocd/scripts/target/` – copy `stm32f1x.cfg` to `stm32f103.cfg`
2. open `stm32f103.cfg` in an editor, search for this line:
    `flash bank $_FLASHNAME stm32f1x 0x08000000 $_FLASH_SIZE 0 0 $_TARGETNAME`
3. replace with:
   `flash bank $_FLASHNAME stm32f1x 0 0x20000 0 0 $_TARGETNAME`
4. in your projects `openocd.cfg` file, change the target to your newly saved target definition:
   `source [find target/stm32f103.cfg]`
5. re-run openocd, flashing should succeed now:

```
❯ openocd
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : J-Link V10 compiled Mar 21 2019 15:43:57
Info : Hardware version: 10.10
Info : VTarget = 3.296 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x1ba01477
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080032ac msp: 0x20002800
Info : device id = 0x20036410
Info : ignoring flash probed value, using configured bank size
Info : flash size = 128kbytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080032ac msp: 0x20002800
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080032ac msp: 0x20002800
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

closed 2023-03-18 by chtaube

deividAlfa, 2023-03-18:
Thanks. Actually I never used OpenOCD, was submitted by a user.

clytle374, 2024-02-08:
I got bit by this one tonight.  Changes from chtaube worked.  

It did terminate in this, should I be concerned?
`xPSR: 0x01000000 pc: 0x08004150 msp: 0x20002800
** Programming Started **
Warn : Adding extra erase range, 0x08018e70 .. 0x08018fff
** Programming Finished **
`

deividAlfa, 2024-02-08:
If It works, it works, simple as that 👍🏻