deividAlfa/stm32_soldering_iron_controller #51: Hardware design for a universal T12, C245, C210, C115 controller

hattesen, 2023-03-25:
## Objective

I would like to look at the feasibility of producing a soldering station controller capable of controlling Hakko T12, JBC C245, JBC T210, and possibly JBC C115 handsets, and auto-detecting which handset type is connected. I would imagine that C115 handsets could be supported by simply limiting the 24V heater PWM duty cycle to 50%.

Schematics for the KSGER and QUICKO hardware is available, and to simplify porting the firmware, the universal controller should resemble one of those, as far as possible, while providing software controlled switching of the TC sensing, which will be required to support both Hakko T12 and JBC cartridges, and detecting which iron is currently connected.

## Design Goals

My goal is to:

1. Identify the hardware requirements for the controller to allow universal hardset support and auto-detection, including connectors and pinouts.
2. Design and capture the schematics for the universal controller, using KiCAD (6 or 7).
3. Layout a PCB using KiCAD, to either fit some standard enclusure or with a custom (FreeCAD) 3D model for 3D printing.
4. Have a few prototype PCBs fabricated at JLCPCB or PCBWay, possibly including assembly.   
   External connectors for 24V DC supply, standby-handset sensor (JBC) and ground/earth would be provided.

To achieve handset-type detection, the handset connector would either have to be specific to this controller (not ideal), or multiple connector should be provided/supported for the different handsets, allowing handsets to be purchased and used directly. If multiple handset connectors are available, it should be possible to detect if multiple handsets are simultaneously connected, and if so, display an appropriate error message.

Possible future design enhancements: USB-C PD connection providing (20V) power and firmware upgrade. This could influence the choice of MCU to possibly use the STM32G0B1/C1 which has USB PD support built-in.

## Project Setup

The design documentation, schematics and PCB layout would be available in a (GitHub) repository, which I would be happy to create and maintain. However, before I do so, I would like to hear if someone else would like to participate in generating the design requirements, BOM and handset connector interface to support auto-detection and safety. Make yourself known if you believe you are willing and able to contribute to this phase, and if a few people come forward, I'll create the project.

I imagine that the firmware could be developed and tested in a feature branch of a forked [stm32_soldering_iron_controller](https://github.com/deividAlfa/stm32_soldering_iron_controller) repository, allowing it to be merged back into the main project without causing any instability and "noise" until then.

Any comments/opinions on this endeavor are welcome and appreciated :)

deividAlfa, 2023-03-25:
I'm not currently interested, have little time for this kind of stuff, also it would be mostly reinventing the wheel.

If you're doing your own design, why keep the limitations from ksger, etc?
Just use a larger 6/7/8 pin GX12 connector, take one pin as handle detection, using a specific resistor for each handle type, then read it with the ADC.

hattesen, 2023-03-25:
@deividAlfa, I don't really feel it is "reinventing the wheel", as all current hardware designs require hardware mods on the controller to change between Hakko and JBC handsets due to the different TC connections.

If I design a "universal" controller, I would like users to be able to purchase and plug in a handset (T12, T245, T210) directly, without having to mess around with rewiring the plug.

T12 option is attractive due to the high availability of after market (clone) handsets and cartridges.
JBC T210 and T245 handsets and cartridges are generally higher quality and offer better ergonomics than the T12, albeit at a higher cost.

And I believe the changes required to the firmware would be limited and confined, and therefore easy to keep up to date with the main project.

deividAlfa, 2023-03-25:
The only hardware mod you need is to split sensing and power pins, then joining them at the T12/C210 handle connector or using it independently in the C245.
It takes 5 minutes, then simply swap the profile in the fw when changing the handle type.
If you want automatic detection, some additional mod would be required.

Of course, it's great if you have the willing and time to make a new design, but I can't help in making it happen else than short comments, spending large amounts of time in this is not possible.

hattesen, 2023-03-25:
@deividAlfa I wouldn't expect you to spend any time on my little pet project, and I will be quite capable of forking and amending the firmware to support auto-sensing the handset type.

deividAlfa, 2023-03-25:
What I mean is, if you're doing all this *just* for automatic handle detection, then better make a simple mod instead doing the entire thing?
Swapping the T12 connector with other having one more pin should be enough.
Cheers!

AndyC52, 2023-03-26:
I recently started a similar, but simpler project. In an attempt to learn KiCAD, i made a schematic based on KSGER designs and a rough PCB layout. I got an estimate from JLCPCB, 12-15€/board. (Excluding display, encoder, shipping and taxes)
The main reason is of course to learn something. I'm reinventing the wheel, but I'm still learning something along the way.
Some of my ideas were to simply make a drop-in module compatible with existing enclosures, but also perhaps boards designed for the larger LCD and OLED screens. 
This firmware is awesome and I think it's worth putting some effort into hardware design as well.
Some hardware improvements can be made while still keeping compatibility with the firmware as-is.
- 3.3V supply
- MCU decoupling
- Op-amp selection
- Spike protection/clamping diodes
- PCB quality
- ...etc.  

When it comes to autosensing _without_ modifying the handpieces, that will take some work. Beyond my  capabilities right now. I have a controller with C245/T12 compatibility, the only mod needed was to join T+/TC in the handle. Works very well.
I don't know how much I'm able to contribute, but an open source hardware for this firmware would be really nice.
Maybe we should start small, start a thread over on the forum to check if anyone else would be interested and able to contribute?
And last, remember the DIY saying: "Why buy something for $10 when you can make it yourself for $150?" :)

hattesen, 2023-03-29:
@AndyC52 thank you for your comment – very interesting to hear about your thoughts.

I don't think the automatic handpiece detection will be a big issue. Hardware changes will require one or two small signal MOSFETs to switch the OpAmp input to the relevant Thermo Couple connections and a signal allowing the MCU to detect handset type and JBC ID resistor value. With regards to firmware support, it should be a fairly minor change once the hardware suport is in place.

As @deividAlfa commented, one way to support both Hakko T12 and JBC handpieces without having two different connectors on the controller would be to provide a single connector (GX12, ideally compatible with current KSGER design), and design an adapter cable to allow OEM JBC handpieces to be connected without modification.

My "stretch goals" for the controller are:
1. USB-C PD power supply support (20V, 60-100W).
2. Support for connecting JBC desoldering tweezers, e.g. [PA120 - Micro Tweezers - JBC Tools](https://www.jbctools.com/pa120-a-micro-tweezers-product-49.html).  
   These tweezers contain two C210 cartridges, connected via the same JBC 6-pin connector as the regular JBC soldering irons. I believe that the two cartridges are connected to power via oppositely polarized diodes, which requires AC power to the heating elements, which would require changing the single MOSFET heater switch into 4 MOSFETs in an H-bridge configuration.

My main reasons for creating a (new) hardware design are:
* Having an Open Hardware controller design for the Open Source [stm32_soldering_iron_controller](https://github.com/deividAlfa/stm32_soldering_iron_controller) firmware.
* Having a single hardware design that allows connecting a wide range of OEM or Clone handpieces with lots of tips/cartridges to choose from – WITHOUT having to make individual hardware mods.
* Allowing makers (and Chinese clone makers) to create new controller variants.
* Externalizing the power supply to allow powering the controller via:
  * Laptop SMPSs (typically 19-20V)
  * USB-C Power Supplies (20V)
  * Laboratory power supplies (24V)
  * Li-Ion Power Banks (USB-C)
  * Power tool batteries (typically 18V Li-Ion)

If you are interested in teaming up in working on...
* Functional requirements
* Enclosure design
* Hardware specifications
* Hardware design and schematics
* PCB Layout
* Prototype production
* Firmware changes

... let me know, and we can join forces on a common (GitHub) project.

AndyC52, 2023-03-30:
Well then, set up the repo, start a thread over on eevblog forum and we'll see where this goes.

closed 2023-04-09 by deividAlfa