deividAlfa/stm32_soldering_iron_controller #52: ksger with c245

labotecno, 2023-04-02:
hello i need a shematic to connect c245 with ksger v2 [https://github.com/deividAlfa/stm32_soldering_iron_controller/blob/master/BOARDS/KSGER/v2/Schematic/Board.jpg](url)
can you help me ?

deividAlfa, 2023-04-02:
Look here:
https://www.eevblog.com/forum/reviews/stm32-oled-digital-soldering-station-for-t12-handle/msg4735031/#msg4735031

labotecno, 2023-04-02:
thank you , and the connection from gx12 ? i have 3 wire from a case toor T12-11 and i want to adapt [image]

deividAlfa, 2023-04-04:
Keep reading that eevblog thread, there's plenty of information in the last pages.
Keep in mind this is not free personal support, only for firmware issues, please use the forum!

labotecno, 2023-04-05:
every thing is ok now you can close this thread

closed 2023-04-05 by labotecno