deividAlfa/stm32_soldering_iron_controller #58: help again handskit

ascarons22, 2023-06-14:
Install the firmware and everything is perfect, but I realize that the handskit pinout is different, it is also 4 wires and it does not have ntc and that is where I need help, I have tried and tried and I have no balls to find the correct combination, I also I have reviewed the 139 pages of eevblogn about your firm and either I don't see it or I'm an asshole, could you please help me. Greetings from Bilbao.

deividAlfa, 2023-06-14:
Haha no problem, I don't have a Handskit myself, but I've seen people with them there, just post a message at eevblog, better with detailed pictures, you'll get assistance for sure!

ascarons22, 2023-06-14:
ok,lo voy a publicar ahora mismo,un saludo

closed 2023-06-17 by deividAlfa