deividAlfa/stm32_soldering_iron_controller #64: Not sure what I'm doing wrong.

Doyle4, 2023-09-02:
I compiled a file yesterday fine with Building_script.bat, I had to choose my board (3) then option 1 to copy files, re-open the .bat and then select 3 and 3 again which worked fine. Today when I run the .bat file it says The screen and STM name then shows a red error - 1 but can't find what the error is.
I deleted the Source folder and unzipped a new one and tried again with same issue.
I have tried copying the 4 files from the V2 folder needed and paste at root, same error, I have also tried without moving the files and relying on the .bat file to move with same error.
Im not very good with coding so I apologise, like I say, yesterday built one fine and flashed, today no luck.

Any help/tips be great.
Love the firmware.

Thanks.

deividAlfa, 2023-09-02:
Sorry I can't help much.

 - Only use CubeIDE 1.12.1, nothing else. Don't have multiple versions installed.
 - The procedure is simple, just download the Code and run building script, choosing your board and then the copy/run MX/compile option.
 - If you're not modifying anything, then simply download a release binary.

Doyle4, 2023-09-02:
Thanks for replying.
CubeIDE 1.12.1 installed but also Utility for flashing the file to STM32.
That was the procedure I did first time to work but no longer does, unmodified source/modified source.
I changed the boot_screen.c, using an online converter to change image to code.

Thanks again.

deividAlfa, 2023-09-02:
Doesn't work even when deleting the entire folder and using a new base?

Doyle4, 2023-09-02:
Tried with deleting folder, and tried with re downloading source.
Will take a good look again before coming back, thanks again :)

Doyle4, 2023-09-02:
Tried making a new user and still no luck, uninstalled CubeIDE, rebooted, reinstall, no luck.
Im going to give up trying as I have no clue what I have done wrong lol.
THanks for your help though!

deividAlfa, 2023-09-02:
What happens when you choose only option 1?
Then import the project into cubeIDE as shown in the video.

Doyle4, 2023-09-02:
Not currently at my Windows machine, but will try this when I can and update.
English (International I think, Im in UK and never choose the US English).

deividAlfa, 2023-09-02:
Hmm now it's also failing for me!

deividAlfa, 2023-09-02:
Fixed!
For some reason CubeIDE rejected to build it, arguing the project already existed in the workspace, though it didn't!

Delete C:\Users\YOUR_USER\STM32CubeIDE\workspace*

Then run the script, should work now.

Doyle4, 2023-09-02:
Amazing, will try this and fingers crossed confirm also.
Once again, thank you!

Doyle4, 2023-09-02:
Amazing! Worked for me also.
Thanks for your time and help.

deividAlfa, 2023-09-02:
Nice! Closing the issue as it's solved!

closed 2023-09-02 by deividAlfa

Doyle4, 2023-09-02:
[image]
[boot_screen.c.zip](https://github.com/deividAlfa/stm32_soldering_iron_controller/files/12504908/boot_screen.c.zip)
Custom Boot screen if anyone else would like it, not to everyone's taste I know :)

deividAlfa, 2023-09-03:
Haha nice!

synthkeywizard, 2023-09-03:
> [image] [boot_screen.c.zip](https://github.com/deividAlfa/stm32_soldering_iron_controller/files/12504908/boot_screen.c.zip) Custom Boot screen if anyone else would like it, not to everyone's taste I know :)

Can you make a photo how it looks at your station?

Doyle4, 2023-09-03:
A soon as I can I will.

Doyle4, 2023-09-03:
[image]

deividAlfa, 2023-09-03:
In white: 

[image]

Doyle4, 2023-09-03:
Prefer it in white, nice!