deividAlfa/stm32_soldering_iron_controller #65: Not an Issue, Don't know how to message you sorry.

Doyle4, 2023-10-25:
Hi man, I looove your KSGER Firmware.

I was wondering if this device has same chipset and possibly could be hacked? Id love to have a CFW on this.
I apologise for asking your issues page.

This device:
https://www.ifixit.com/Guide/XMAX+V3+Pro+Disassembly/157413

deividAlfa, 2023-10-30:
Sorry, this project is not in active development anymore, only ocassional features are added, or small bugfixes.
But that device wouldn't fit here either, this firmware is specifically made for T12 stations, which share a very similar hardware, the one you posted is vaper.
Just because it has a STM32 means nothing, it's like saying two cars are the very similar same because they run on same fuel.
Closing!

closed 2023-10-30 by deividAlfa

Doyle4, 2023-11-04:
No problem dude, Thought id ask incase.
Wish I had the knowledge to tinker with it myself.

Thanks for getting back to me though :)