deividAlfa/stm32_soldering_iron_controller #67: Add support for ksger v3.3 board

LynxChaus, 2023-10-31:
Please add support for v3.3 board:

CPU: STM32F103CBT6

OLED display connected to PINs: 25 (PB12) SCL, 26 (PB13) SDA
Encoder - 38 (PA15) KEY, 39 (PB3) Pin A, 40 (PB4) Pin B
UART? - 42 (PB6), 43 (PB7) (unpopulated 3-pin header near buzzer)
10 (PA0) - SCL (Pin1 IC CIEAH marking, some AT42CXX in SOT23-5 package)
11 (PA1) - SDA (Pin3 IC CEIAH marking)
Tip Temp - 12 (PA2)
Buzzer - 13+14+15 (PA3+PA4+PA5)
Shake - 16 (PA6)
NTC - 17 (PA7)
Heater PWM - 18 (PB0)
VIN Monitor - 19 (PB1)
[image]

deividAlfa, 2023-10-31:
This seems normal v2 hardware.

czullo, 2023-10-31:
I can confirm that this is V2 hardware, i got same on my Quecoo T12-955 and im running on this KSGER_v2_OLED firmware with no problems.
[image]

deividAlfa, 2023-10-31:
Closing then!

LynxChaus, 2023-10-31:
Mention this board in docs, please.

deividAlfa, 2023-10-31:
No need to, this simple rule still applies:
- 4 pin oled: V2.
- 6-7pin oled (Except old Quicko) : V3.

Simple as that!

LynxChaus, 2023-10-31:
Simple for you, who know schematic of gazzilon this boards. For someone, who bought this board - find in $(searchengine) $(boardname) is better than tracing connections from CPU to peripherals. For me this cost is two hours. Grr.

LynxChaus, 2023-10-31:
Well, flashed KSGER_v2_OLED, it's run but report "VOLTAGE LOW" and "NTC READ LOW". 
[image]
What I'm doing wrong?

deividAlfa, 2023-10-31:
- Power it with the 24V PSU, not from the ST-Link.
- NTC is in the handle, connect it.

deividAlfa, 2023-10-31:
Will try to add more board pictures, but last time I has some issues and wasn't able to do so, GitHub blocked the uploads.

LynxChaus, 2023-11-01:
Yes, my bad, I'm run it from ST-Link power. 
Powering from 24v power supply is properly report input voltage, but NTC read -5.0 degrees. Switching NTC detect ON/OFF from Iron menu don't help. Only set manually resistance to 10K and enable "NTC detect" - show temperature                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               [... cut ...]                                                                                                                                                                  +38.x degree (wrong). Switching OFF "NTC detect" - show temperature -99.9 degree.

Well, found inadequate 470K NTC in handle, replaced it to 10k NTC ... and temperature now +5 degree, when NTC resistance is 11.08k (+22 degree). Playing with beta value don't help. Switching PullUp to Down - show temperature +44 degree.
What is wrong here?

deividAlfa, 2023-11-01:
Leave 10K value for NTC. It's pullup, 4k7.
Or simply disable NTC so it uses internal stm32 sensor.

LynxChaus, 2023-11-02:
Aha, understand. Reverted back pull resistance to 4.7k, adjusted beta and now it show normal temperature.
Thnx!

LynxChaus, 2023-12-07:
Great. Write versions (2.1s, v3.3 and so) after "Ksger v2" - for search engines.

deividAlfa, 2023-12-07:
Nope, people must read the Readme!

deividAlfa, 2023-12-07:
Closed, not an issue. Just read!

closed 2023-12-07 by deividAlfa