deividAlfa/stm32_soldering_iron_controller #68: More convenient control for standby/sleep via an encoder.

vlk3ru, 2023-11-12:
Dear author, your software is excellent. I flashed it for my station KSGER  PCB hw: Ve2.12S. But it would be very convenient to send the station to Stanby/Sleep mode by simply pressing the encoder key instead of showing the heating graph. Can you make this optional setting ?

deividAlfa, 2023-11-12:
Good idea. I barely have any time to spend here, today was a lucky day!
https://github.com/deividAlfa/stm32_soldering_iron_controller/releases/tag/v1.10.6

closed 2023-11-12 by deividAlfa

vlk3ru, 2023-11-13:
I updated my KSGER station to 1.10.6 firmware. Everything works great. Management has become very convenient. Thank you very much !

deividAlfa, 2023-11-13:
Enjoy!

Doyle4, 2023-11-22:
Amazing! I was looking for an option to do this, make sense now why I couldn't find it. 
Thanks :)

synthkeywizard, 2023-11-22:
If it was user-configurable from setup menu - it would be a great improvement.

Doyle4, 2023-11-22:
Love this firmware, but I couldn't get use to putting in standby/sleep clock rotating, would either go to graph or lower temp haha, this is a huge improvement for me!

deividAlfa, 2023-11-22:
Sorry, no option will be done, this will be the new setting from now!
The graph screen is absolutely useless, only a fancy thing, so it should receive the least quick access.
Setpoint adjustment, sleeping and menu are the top priority actions, so they hot the easiest gestures.

serisman, 2023-11-22:
Thank you for making this change!  This fixes one of minor annoyances I was having with this excellent firmware.

One thing I noticed is that I have to rotate the encoder to exit sleep mode instead of just pressing the button.  Is this on purpose, or an oversight?

deividAlfa, 2023-11-22:
Yes, accidental clicking would be a mess, resuming run mode.
You have to rotate only one step, so the setpoint won't even be shown.

serisman, 2023-11-22:
Ok, that makes sense.  Thanks for the response.

Doyle4, 2023-11-24:
Installed and working as stated by others, great update feature!