deividAlfa/stm32_soldering_iron_controller #71: Question not an issue.

Doyle4, 2023-12-09:
Im still currently on 1.10.6 and was about to update to 1.11.0 and spotted a newer update today.

1.11.0 mentioned it will wipe all data which is fine, but 1.11.1 says data will stay, is this only for people who update from 1.11.0 to 1.11.1 or will updating from 1.10.6 > 1.11.1 keep my current settings? 
Like I say, I honestly don't mind it wiping, just wondering before I do the update and to have tip calibration tool at hand for when I do update.

Thanks again!

deividAlfa, 2023-12-09:
It clearly says ”It won't wipe data coming from 1.11.0". Only from 1.11.0.

closed 2023-12-09 by deividAlfa

Doyle4, 2023-12-09:
I apologise for having learning difficulties and not understanding it properly, Jeez....