deividAlfa/stm32_soldering_iron_controller #74: Profile for AD597 + GOOT RX-80HRT ?

vlk3ru, 2023-12-16:
I use KSGER V2.1 stations with an AD597 amplifier and GOOT RX-80HRT+RX-85HRT tips with a K-type thermocouple inside these cartridges. Which profile should I use out of the three Hakko T12 / JBC C210 / JBC C245 ?

deividAlfa, 2023-12-16:
This is not an issue, use eevblog forums in the future.
How would I know? This nowhere standard, probably none!
The default amplifier gain in these stations is about 250, the max output is 3.3V, so the it will clip at 13mV input.
K-type outputs 20mV  @ 500°C,  12mV @ 300°C, your will have to reduce the gain of the amplifier to the half by replacing the 249K resistor with 120K.

Then choose any profile, open calibration menu, insert tip now (Not before), open settings and manually adjust 250 and 400°C steps while measuring the tip temperature.
Then save and run the automated calibration.

closed 2023-12-16 by deividAlfa

vlk3ru, 2023-12-16:
I immediately removed the stock amplifier in 2019 and instead installed a custom AD597 thermocouple amplifier and a goot pen with RX-8xHRT cartridges. This all works great with KSGER firmware 2.10 and even better with your firmware 1.10.6. Everything is fine. Thank you for your firmware. My question is different: Which profile should I choose in your firmware? There are only three profiles in the firmware. And my case is not. Maybe it would be better to add a fourth user profile? Having the name "USER" for example ? It's just a question.

vlk3ru, 2023-12-16:
[image]

My diagram is for clarity.

vlk3ru, 2023-12-16:
Question about this:

gui_strings.c :
char * const profileStr[NUM_PROFILES] = { "T12", "C245", "C210" };

Do you think it is possible to add a fourth profile ?

deividAlfa, 2023-12-16:
You can use any. Only the default calibration changes, which can be changed, so there's no difference except in the name.

You can rename the profile name in:
https://github.com/deividAlfa/stm32_soldering_iron_controller/blob/master/Drivers/graphics/gui/screens/gui_strings.c#L1303

The default tip names are here:
https://github.com/deividAlfa/stm32_soldering_iron_controller/blob/master/Core/Src/settings.c#L633

The default calibration is in your controller type profile, BOARDS/KSGER/x/STM3wF1xx/Core/Inc/board.h

Modify as required and compile for your own.

vlk3ru, 2023-12-16:
Thanks for the answer.

deividAlfa, 2023-12-16:
Powering the amplifier with 24V is dangerous,  
if anything goes wrong, even with the resistor divider, the stm32 will get 15V.

Put a 3.3V zener in parallel with the 1K8 resistor!

vlk3ru, 2023-12-16:
Thank you very much !
I will definitely take your recommendations into account and change the circuits of my devices.

deividAlfa, 2023-12-16:
Also, there's no memory left in 10KB devices (STM32F101), so another profile would not fit.

vlk3ru, 2023-12-16:
Ok.
In addition to the 3.3V zener diode, according to the AD597 datasheet, +5V power supply is possible instead of +24V.

vlk3ru, 2023-12-16:
> Also, there's no memory left in 10KB devices (STM32F101), so another profile would not fit. A STM32F103 (20KB) would, but I have to keep compatibility with all.

I just used Profile T12. Just without thinking. But apparently it is possible to use any profile for calibration and work in my case.