deividAlfa/stm32_soldering_iron_controller #75: Question: FW compatibility

vookimedlo, 2023-12-16:
Hello,

I have just bought the KSEGR T12 and have the following controller board inside. I cannot find this board layout on any forums so I ask directly. Can I use your FW with this board?

Thank you.

[image]

vlk3ru, 2023-12-16:
This is probably KSGER HW version 2.0 .

FW : https://github.com/deividAlfa/stm32_soldering_iron_controller/releases/download/v1.11.2/KSGER_v2_OLED.zip

[image]

vlk3ru, 2023-12-16:
[image]

deividAlfa, 2023-12-16:
Another V2 yeah. Read the Readme for the 64kb issue.

vookimedlo, 2023-12-16:
Thanks, will try to replace the MCU with STM32F103CBT6.

closed 2023-12-16 by vookimedlo

deividAlfa, 2023-12-16:
There's no need to replace it, read [`Compatibility`](https://github.com/deividAlfa/stm32_soldering_iron_controller?tab=readme-ov-file#compatibility)!

vlk3ru, 2023-12-17:
> There's no need to replace it, read [`Compatibility`](https://github.com/deividAlfa/stm32_soldering_iron_controller?tab=readme-ov-file#compatibility)! 64KB devices actually have 128KB!

Sorry, I didn't know about this.