deividAlfa/stm32_soldering_iron_controller #108: The sound is the same for boost, standby, sleep modes.

vlk3ru, 2024-02-16:
The sound is the same for boost, standby, sleep modes. Sometimes, when working without looking at the display, the same sound leads to errors. Dear author, could you make them different?

deividAlfa, 2024-02-17:
Not sure if this subject deserves any effort? It only needs a quick peek to the display when it beeps.

vlk3ru, 2024-02-17:
Example:
boost - long sound it's high temperature.
standby - short sound
sleep - two short sound.

deividAlfa, 2024-02-24:
That's what cheap encoders do, I can't do anything

vlk3ru, 2024-02-24:
I ordered this type of encoders, but they are even tougher. :(

deividAlfa, 2024-02-24:
Don't buy Aliexpress encoders! Genuine Bourns are cheap enough, $2 or less!
I made a workaround, si if a click is issued very quickly after iron working changed, it will revert the working mode and enter sleep.

deividAlfa, 2024-02-24:
Try v1.13.7

closed 2024-02-24 by deividAlfa

vlk3ru, 2024-02-24:
Ok. Thanks you.