deividAlfa/stm32_soldering_iron_controller #111: porting to Handskit Mini Rework Station

Careless, 2024-02-22:
is there the possibility to port this firmware over to a Handskit Micro Rework Station firmware?

https://www.aliexpress.com/item/1005002890887241.html

It uses the same interface and I believe it uses a similar STM32 LCD expansion board for the display and control software, but the calibration is limited and requires an external temp sensor to dial in the hot-air temp accurately.

It only needs a software mod to be a much better portable 868D based rework kit.

deividAlfa, 2024-02-22:
I don't have the hardware, I can't.

deividAlfa, 2024-02-22:
I'm fine with my 858D!

deividAlfa, 2024-02-24:
Closed, I will not buy new hardware to support elses's station. If you wish support, consider donating.

closed 2024-02-24 by deividAlfa