deividAlfa/stm32_soldering_iron_controller #131: Bootloop KSGER-T12 V3.3

jporte, 2024-05-29:
I have a problem with my KSGER-T12 V3 soldering station.

I have downloaded the appropriate firmware ‘KSGER_V3_OLED_Release.bin’ to the soldering station. Unfortunately, it now keeps restarting when the boot logo is displayed. A bootloop, so to speak. 

I have also tried older versions. Sadly without success.

I have this board: https://github.com/deividAlfa/stm32_soldering_iron_controller/blob/master/Readme_files/boards/ksger_v3_3.jpg

Has anyone had the same problem and been able to find a solution?
The microcontroller seems to be an original STM32F103RBT6.
I have already worked according to the ‘Clone detection’ instructions. This also showed that it should not be a clone.

I look forward to constructive answers.

deividAlfa, 2024-05-29:
Can't tell. What are the values of DBGMCU_ID and Flash Size register?
Try erasing the whole chip, also check the option bits and make sure there're no protected areas.

I guess it worked correcttly in the initial setup screen and started crashing afterwards?

jporte, 2024-05-30:
Thank you for your answer.

STM32 ST-LINK Utility under Windows says:
ID: 0x410
Flash size: Unknown

The output of your test programme is:
CPUID: 411fc231
Flash: ??? kB
Theoretical: ???
Detected: 1024 B

I overlooked the part with the flash size in the ‘clone-detection’ instructions. There I get a value of FFFFFFFFFF.

So it seems to be a fake chip after all. I will replace it with an original when I get the chance. Hopefully it will work then.

Sorry for opening this issue. If I had read more carefully, I wouldn't have had to bother anyone with it.

closed 2024-05-30 by jporte

deividAlfa, 2024-05-30:
I'll make a test build forcing 1KB flash page, let's see if it works this way.
Will update in a few hours.

deividAlfa, 2024-05-30:
Try this. But will probably fail anyways.
[STM32SolderingStation.zip](https://github.com/deividAlfa/stm32_soldering_iron_controller/files/15505948/STM32SolderingStation.zip)

jporte, 2024-05-31:
I have tried it out. Unfortunately, that didn't work either.

After I flashed the programme, I tried to download it again with the flash size as a parameter.

st-flash --flash=64k write STM32SolderingStation.bin 0x08000000

I got this error:
st-flash 1.7.0
2024-05-30T23:51:05 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 65535 KiB flash in at least 1 KiB pages.
Forcing flash size: --flash=0x00010000
file STM32SolderingStation.bin md5 checksum: ca1faa87b6859fa51da955cc5b1e81e, stlink checksum: 0x00b1e5aa
2024-05-30T23:51:05 INFO common.c: Attempting to write 111876 (0x1b504) bytes to stm32 address: 134217728 (0x8000000)  
2024-05-30T23:51:05 ERROR common.c: addr too high
stlink_fwrite_flash() == -1

After that, however, I got past the boot screen.
This can also be replicated by executing the command when the boot screen is displayed.

Unfortunately, it no longer worked after a restart and I got a bootloop again.

I can't avoid soldering in a new STM32. But that's not a problem either.

I'll keep you up to date. But it may take a few weeks.

deividAlfa, 2024-05-31:
Wait, why are you using `--flash=64k`?
Should be `--flash=128k`

jporte, 2024-05-31:
Yes, I downloaded the programme to the STM32 with --flash=128k. As this was unfortunately still unsuccessful, I tried it with 64k just for fun.