deividAlfa/stm32_soldering_iron_controller #140: Fixed room temperature.

porlock, 2024-12-22:
deividAlfa, does room temperature is taking into account in calculations for tip temp ? if yes it would be beneficial to add an option manually set room temperature in firmware, my NTC change in -+ 20C depends on electronic heat up, but my actual room temperature is always in 5C change range, i i could just set it fixed as 24C and error would always less than 3C. Not like now .. when using NTC measurement it can change from 20C co 40C.

deividAlfa, 2024-12-22:
It's not room temperature, but the cold side of the tip, normally in the handle, where the thermocouple ends.
Search cold junction compensation.

closed 2024-12-22 by deividAlfa

porlock, 2024-12-22:
Yes I know but it's actually equals  to room temperature, and for example jbc don't have sensor in handle,and there is an option to connect  NTC senator directly on controller board .

deividAlfa, 2024-12-22:
I repeat it's not the room temperature!

porlock, 2024-12-22:
Yes, when I installed ntc on control PCB it's more or less correct when I turn it on, like 22C after 30min it's 45C