deividAlfa/stm32_soldering_iron_controller #151: Does the ksger v2.1S work?

OpenCoreMicha, 2025-07-25:
I just bought a ksger v2.1s with stm32 not v3.1S bc all of those have since 2021 only have mm32 will it work?

OpenCoreMicha, 2025-07-25:
I dont have it i ordered it today and it will take some time

OpenCoreMicha, 2025-07-25:
If i have i Could check

OpenCoreMicha, 2025-08-04:
Hello does it work or no @deividAlfa

OpenCoreMicha, 2025-08-04:
Welp gonna close it now reopen it if you got the answer

deividAlfa, 2025-08-04:
Do you know I'm a person that works, eats, sleeps and all that?
I'm not a company support center, I answer when I can...

Meanwhile make the effort of disassemblying the board and making proper pictures, all I can see is a stm32, but the board layout could be different.

Still, it seems a V2 variant.

niobebo, 2025-08-09:
@OpenCoreMicha 

the PCB looks very professoinal   
Which retailer is this Device from?

Do you have a link?

OpenCoreMicha, 2025-08-09:
Yes I got it from aliexpress here  is the link I bought it from
https://a.aliexpress.com/_EJwVhom

OpenCoreMicha, 2025-08-10:
thanks will be  buying a st-link now